dan-fritchman/Hdl21 #1: Python-Style Slicing & Concatenation

dan-fritchman, 2021-07-21:


aviralpandey, 2021-07-26:
A basic version of this exists in the branch `python-style-slicing-1`. 

One possible thing to consider is supporting stepped slicing. While not supported by Verilog, I believe the functionality is generally useful. Do we want to include this in the proto?

dan-fritchman, 2021-07-28:
Updates look great. 

On stepping: I would probably recommend holding off, until finding a use-case we find compelling. Like you said it's an unconventional feature in the legacy HDLs. And it's a convenience feature here; we can always create them via concatenations like 

```python
# For step=2
h.Concat(sig[0], sig[2], ...)
```

There's also the related question of whether `step`s would be first-class supported in the serialization schema, or just helpful syntax in Python. For the former, the only seeming schema addition would be a `step` attribute on `Slice`:
https://github.com/dan-fritchman/Hdl21/blob/acefe0b93788e421dcaa74e8f815f32c1fb40775/hdl21/proto/circuit.proto#L55

Nonetheless I'd again advocate holding off. 

--- 

We can review edits in this morning's meeting if you like; if not I'm happy to get em merged.

dan-fritchman, 2021-09-28:
Seems we didn't quite get this as we'd like. 
@aviralpandey planning some further changes on this front.

dan-fritchman, 2021-10-30:
Commit dac6c76 has a mostly-fleshed-out version of this. 

Writing it makes me realize our proto-buf-schema has no facility for bit-wise reversal (like verilog's `upto` vs `downto`, or `[5:0]` vs `[0:5]`, without resorting to bit-by-bit concatenations. The `Slice` message fields are: 

```
// Signal Slice
message Slice {
  // Parent Signal Name
  string signal = 1;
  // Top Index
  int64 top = 2;
  // Bottom Index
  int64 bot = 3;
}
```

Proposal:

* Change `top` and `bot` to `start` and `stop`, with the semantic-level meaning similar to the Verilog `[5:0]` and `[0:5]` indices - `[start:stop]`.
* Map Python-level `Slice`s such that, for a 6-bit `Signal`: 
  * `step is None` (the default) or `step == 1` <=> `[5:0]`
  * `step == -1` <=> `[0:5]`
  * All other values of `step` are converted to `Concat`s

The latter proposal is not implemented.  
There is an `xfail`ing unit test which exercises these export cases.

ThomasPluck, 2023-07-09:
This is also cropping up as I'm just reaching for certain Pythonic syntax that'll throw. In particular, I'm hoping for symmetric slicing, eg.:
```python
a[1:3] = [1,2] # This does not throw
```
So, this is just a quick change in the definition of `sliceable` with lotsa downstream effects. Looks like the bulk of this and #3, will be convincing elaboration to let it go.

closed 2024-03-21 by dan-fritchman