dan-fritchman/Hdl21 #3: Connection Rules to Instance-Array Elements and Slices

dan-fritchman, 2021-09-28:
One of the next features on deck will be connecting to individual `Instances` (and perhaps slices thereof) from `InstanceArrays`. For example: 

```python
@h.module
class M:
  a, b, c = h.Signals(3)
  arr = 5 * OtherM()  # Creates an array 

  # Now connect to its individual Instances
  arr[0].some_port = a
  arr[1].some_port = c
  arr[2].some_port = b
  # Etc
```

This would frequently be a helpful feature, and one which popular HDLs generally lack. 

But it does generate a few policy questions for what constitutes a legal connection. And they're not as easy to concisely state & implement as for (scalar) `Instances`. 

Primarily, what happens when an `Instance` gets *both* such a "slice connection" and "array connection". As in: 

```python
@h.module
class M:
  w5 = h.Signal(width=5)
  other = h.Signal(width=1)

  arr = 5 * OtherM(some_port=w5) # Creates an array, connects a bit of `w5` to each `some_port`
  arr[0].some_port = other # Then try to connect a scalar Signal to one of its underlying Instances
```

There are a few policies I can imagine: 

* **Error Time**. Somewhat ambiguous intent such as this generates an error during elaboration. 
* **"Last-In Wins"**. Whatever connection-calls are made to whatever layers of array and instance hierarchy, the final one dictates the ultimate connections. 
* **Tiered Rules**. Some prioritization of either individual-instance or array-level connections. Probably prioritizing "more specific" connections, perhaps with a warning message when over-riding less-specific ones. 

## Last-In 

For scalar `Instances`, "last-in wins" is the effective (and I think, best) policy. For example this works just fine: 

```python
@h.module
class M:
  some = h.Signal()
  other = h.Signal()

  inst = OtherModule(some_port=some)  # Initially connect to `some`
  inst(some_port=other)  # Changed our mind, connect to `other` instead
```

Here the Port `some_port` on Instance `inst` is ultimately connected to the Signal `other`. The short-lived connection to Signal `some` is effectively ignored. 

The implementation of "last-in wins" is both implicit and straightforward: each `Instance`'s connections are kept in a dictionary. Writing new connection key-value pairs just replaces the old ones. 

For arrays this wouldn't quite work. Taking the example above: 

```python
  arr = 5 * OtherM(some_port=w5) 
  arr[0].some_port = other  
```

Here `arr` would be an `InstanceArray`, and `arr[0]` would be some other kinda thing yet to be designed. Let's call it `InstanceRef`. The two would need some means to reconcile "last-in" connection ordering. Presumably the array could keep track of this as a time-ordered list of connections - both to itself at large, and to any constituent members. 

## Tiering

I could instead imagine a set of rules favoring "more specific" connections rather than "last" ones. This priority would presumably look something like: 

1. (Highest Priority) Individual `Instance`, or single-`Instance` slice of an array. E.g. `OtherM()`, `arr[0]` above. 
2. Multi-`Instance` slices of arrays, e.g. (theoretically) `arr[0:1]` above. 
3. (Lowest Priority) Complete `InstanceArrays`, e.g. `arr` above. 

I *think* we will always be able to tell which is which. (I don't think, for example, a `Generator` could vary whether it connects to an `Instance` vs a slice across runs.) 

## Other Language Precedents

The popular HDLs (Verilog, VHDL) do not include the kinds of "procedural connections" under discussion here, and therefore do not offer much insight. Their ethos is generally closest to "Error Time", in that connections must be quite unambiguous to be valid. 

I believe Chisel uses the last-in-wins rule in general. I do not know whether it mixes this with anything like the "tiered priority" concept for Instances vs Arrays.

dan-fritchman, 2021-09-30:
Notes from discussion 9/30/21: 

* Probably an overall preference for "last-in wins", among these policies
* Resolution: leave instance-array-slicing unsupported for the time being 
* Sit tight until identifying an application with more benefit from it. 

Also discussed the prospects for *multi-dimensional* instance arrays, a feature lacking in most popular HDLs.  
That probably deserves its own issue.

dan-fritchman, 2023-06-24:
Lots of time has passed since that last comment; no pressing demand for the "connect into instance-array sub-slices" feature has arisen. Resolution NTBF.

ThomasPluck, 2023-07-07:
I've reopened this issue because I do have [a program](https://github.com/ThomasPluck/tt04-biased-trng/blob/d2a4a39bc0c4e4c99abea21054f28992cb8947d0/sim/ogates.py#L121) where I want to use `InstanceArray` this way.

This is my take right now:

- InstanceArray is an Instance + a count that is an elaborated into Module filled with repetitive instances
- We presumably want this to work both pre and post elaboration. So a couple ideas here:
   - Maybe `InstanceArray` could be a `@calls_instantiate` object with `_mult` on a module, actually being a method to call the `InstanceArray`
   -  `InstanceArray` should then have `__setitem__` defined, so that it can create connections to its managed instances.
- I think this will adequately describe what you want until the `proto` pipeline, then there will have to be some logic mimicking `ArrayFlattener` to avoid a double-PR situation again.

dan-fritchman, 2023-07-18:
Great topic to dig into if you find it helpful! 

I don't totally follow the use-case though. I expect you mean this passage: 

```python
    # Generate pairings
    for k, v in pairings.pairings.items():

        temp_name = f"{params.node_names[k[0]]}{params.node_names[k[1]]}_coupling"
        
        ogate.add(
            gen_coupling(
                width=v[-1],
            )(
                # FIXME: Shouldn't there be a better way?
                # A=ogate.arr.osc[v[0]],
                # B=ogate.arr.conns.osc[v[1]],
                VDD=ogate.VDD,
                VSS=ogate.VSS,
            ),
            name=temp_name
        )

    return ogate
```

* What's, like, the ideal API for what you want to do here? 
* Is `Concat` not a good solution for it? 

--- 

As for the implementation in #155 - a big part of the point of `InstanceArray` is to follow "broadcast rules", similar to Verilog, popular schematic environments, and (less similar to, but still similar-ish to) numpy. 

E.g. 

```python
@h.module
class M:
  s = h.Signal(width=2)
  VSS = h.Signal(width=1)
  rs = 2 * h.R()(p=s, n=VSS)
```

After elaboration should produce something where

* `M.rs_0.p` is connected to `M.s[0]`
* `M.rs_1.p` is connected to `M.s[1]`
* `M.rs_0.p` is connected to `M.VSS`
* `M.rs_1.p` is connected to `M.VSS`

Is that the case in the current implementation? It looks like we are broadcasting "the whole" array-width signal to every instance instead. 

--- 

More philosophically - the general MO of much of HDL21, this `InstanceArray` included, is "don't really do much during its API calls; just store stuff to be processed later during elaboration". That philosophy is helpful in many respects, and less so in many others. "Pulling forward" the array-expansion to API-call time should work in general. But there are cases where it won't. Particularly those where other data the array depends upon has not yet been computed - for instance because they are part of `generator` results. E.g. 

```python
@h.generator
def WidthEitherOneTwoFourOrEight(params: Params) -> h.Module:
  m = h.Module(name="m")
  m.p = h.Port(width=either_one_two_four_or_eight(params)
  return m

@h.module 
class Other:
  p = h.Port(width=8)

@h.module 
class HasM:
  o = 8 * Other() 
  m = 8 * M(some_params)(p=o.p)
```

Those widths are chosen so that any value *would* work at elaboration-time - but we don't know which it is, as of the attempt to connect `HasM.o` to `HasM.m`. 

That example may be contrived. But it works, and wouldn't if divying up arrays "right away". (The next thought might be "evaluated `M(some_params)` right away too" - i.e. kick generator-running out of elaboration and into "construction time" too. Separately, that may not be a bad idea. But certainly would be more stuff to do.)

ThomasPluck, 2023-07-18:
> What's, like, the ideal API for what you want to do here?

I just think in terms of a graph with numpy, I have a 2D array of inverters (who's columns are ring oscillators) I want to route my coupling modules to reflect my math.

So something along the lines of:

```python
osc[0].inv[2] = osc[1].inv[3]
osc.VSS = VSS
```

So, I have fine-grained control over what in my array is getting wired, but I also have broadcasted control over generalities like: all ground ports should be grounded.

> Is `Concat` not a good solution for it?

If it isn't in docs or easily understood from obviously task-relevant examples, I think its safe to say that I haven't tried it. So, let me take it for a ride and see if this gets at what I was thinking. BRB

Back, I guess you could make this work by sticking all of the signals in a 1D array with Concat and then just connecting using a row-wise style array access like in C... and then if you wanted you have a handy wrapper class to handle this into higher dimensions...

Okay - there's an idea, so instead of modifying `InstanceArray` from the ground up, you can instead create an "ArrayInstance" which simply wraps all of this behavior around a bunch of concat'd signals that then dissolves at elaboration to leave behind what we know is there the entire time.

ThomasPluck, 2023-07-18:
Pretty happy with my now functional HDL21, using sliceable signals and `Concat` cleverness. Between this dragging up an ancient issue for what really becomes boils down to an indexing convention... I think I'm going to let sleeping dogs lie.

closed 2023-07-18 by ThomasPluck