dan-fritchman/Hdl21 #37: `Signal` Multiplication for Replication

dan-fritchman, 2022-07-27:
I find myself using many "plural" `Signal`, `Port`, and similar declarations.  
Example from our README: 

```python
import hdl21 as h

@h.module
class MyModule:
    a, b = h.Inputs(2)
    c, d, e = h.Outputs(3, width=16)
    f, g, h, i = h.Signals(4)
```

The one facet I've never loved is having to add the "number of signals" argument, as it doesn't immediately appear clear what those numbers (`(2,3,4)` here) should mean.  

Proposal: make `int * Signal` multiplication do essentially the same thing.  
The same result would be achievable with: 

```python
import hdl21 as h

@h.module
class MyModule:
    a, b = 2 * h.Input()
    c, d, e = 3 * h.Output(width=16)
    f, g, h, i = 4 * h.Signal()
```

I think this may be more clear.  
It would work very much like how Hdl21 `InstanceArray`s are typically generated, via `int * Instance`.  

And given Hdl21 has no ambitions to capture behavioral code in which `int * Signal` would produce a _value_ of the Signal instead - e.g. during a "procedural block" - I don't think it would be in the way of anything else.

growly, 2022-08-15:
Are you suggesting to get rid of `h.Inputs(...)` form or to allow both?

```a, b = 2  * h.Input()```

is nice because it's almost exactly the python list syntax,

```a, b = 2 * [h.Input()]```

which would do the same thing without any magic.

What's your philosophy regarding how much of Hdl21 is DSL and how much is Python?

dan-fritchman, 2022-08-16:
Clarifying this note - 

> a, b = 2 * [h.Input()]
> which would do the same thing without any magic.

While I do agree in liking the similarity in syntax, those two code-blocks *do not* do the same thing.  

The latter, list-mult version:
* Creates _one_ `Signal` object
* Creates a list with two references to it
* And then assigns the two references to variables `a` and `b`. 

Similar to if one did: 

```python
x = h.Signal() # Only one `Signal` ever gets created, here. 
two_refs = [x, x]
a, b = two_refs # It just gets two references
```

The `Inputs` function (and `Outputs`, `Signals` et al) do something more like: 

```python
def Things(n: int): 
  rv = []
  for _ in range(n): 
    rv.append(Thing())
  return rv 
```

The difference is solely whether each element is a distinct in-memory object, or a reference to the same one.  

> Are you suggesting to get rid of h.Inputs(...) form or to allow both?

Don't have a strong feeling here. I suppose allow both.

closed 2023-06-05 by dan-fritchman