dan-fritchman/Hdl21 #38: Non `Module` attributes in class definitions

dan-fritchman, 2022-07-31:


growly, 2022-08-15:
It's not clear why one way or the other would be advantageous (or disadvantageous) to a user. Insofar as a DSL is valuable in the first place because it forces you to express only as much and as little as you need, I think there's value in constraining the types of attributes and limiting their visibility.

Whether those attributes outlive "elaboration" might become a matter of convenience: e.g. will users be able to write tests than depend on those attributes?

Whether errors are caught while ingesting a class body or later while doing some "elaboration" step isn't immediately clear. What is the difference to the designer's flow? Are they going to be iterating on describing a class to get it right or are they going to be iterating on their entire HDL description to get it right? I don't know.

(Are other polymorphism-esque conveniences going to be available, like derived classes (Modules)?)

dan-fritchman, 2022-10-26:
Outlook: before #53

closed 2022-10-27 by dan-fritchman