dan-fritchman/Hdl21 #40: What should `NoConn`s be able to do?

dan-fritchman, 2022-07-31:


growly, 2022-08-16:
I agree. Perhaps also create a `NoConn` that always acts as a single-width, and a `NoConns` that conveniently expands to connect however many wires as there are to `NoConn`s? There would be ambiguity in some concatenations, but then the user can fallback to the single-width:

:heavy_check_mark:  `some_concat = h.Concat(some_sig, h.NoConns(), pwr, gnd, another_bus)`
:x: `some_concat = h.Concat(some_sig, h.NoConns(), pwr, gnd, h.NoConns()`
:heavy_check_mark: `some_concat = h.Concat(some_sign, 3 * h.NoConn(), pwr, gnd, h.NoConns())`

dan-fritchman, 2022-08-16:
Agree the `Concat`s are probably the diciest part. 

And a very good point - the Signal-multiplication change proposed in #37 would also be a nice play to specify the "width" of `NoConns` in such concatenations. (Really by just creating several of them.)

ThomasPluck, 2023-01-03:
I've taken a look at the NoConn problem using the example test:

```python
def test_elab_noconn3():
    """Test bundle-valued noconn"""
    
    @h.module
    class M2:
        p = h.Diff()

    @h.module
    class M1:
        m = M2()
        m.p = h.NoConn()

    h.elaborate(M1)

    assert len(M1.signals) == 1
```

The error occurs in PortRefs, with some debugging, I'm noticing that before this ElabPass begins that the Module has an empty Bundle dict. So I'm beginning to think that a Bundle-valued NoConn solution without the rest of the Bundle elaboration plumbing is going to be hacky one.

So, good first step, could be to start by making the Bundles attached to Modules available throughout the elaboration process and then figure out where NoConns feature around that?

dan-fritchman, 2023-01-04:
> The error occurs in PortRefs, with some debugging, I'm noticing that before this ElabPass begins that the Module has an empty Bundle dict.

Not sure I totally follow what you mean here.  
Expanding on that example: 

```python
import hdl21 as h 

@h.module
class M2:
    p = h.Diff()

@h.module
class M1:
    m = M2()
    m.p = h.NoConn()

# Expansion here! 
print("M2.bundles:")
print(M2.bundles)
print("M1.bundles:")
print(M1.bundles)
```

Yields: 

```
M2.bundles:
{'p': BundleInstance(name=p of=Bundle(name=Diff))}
M1.bundles:
{}
```

Which is right - M2 has a bundle-instance, and M1 does not. 

If we do this again after elaborating up to the `RESOLVE_PORT_REFS` pass, which handles `NoConn`s: 

```python
from hdl21.elab.elabpass import ElabPass

h.elaborate(
    M1,
    passes=[
        ElabPass.RUN_GENERATORS,
        ElabPass.ORPHANAGE,
        ElabPass.INSTANCE_BUNDLES,
        ## ElabPass.RESOLVE_PORT_REFS ## This would be next, stop right before it
    ],
)

print("M2.bundles:")
print(M2.bundles)
print("M1.bundles:")
print(M1.bundles)
```

We get the same thing: 

```
M2.bundles:
{'p': BundleInstance(name=p of=Bundle(name=Diff))}
M1.bundles:
{}
```

---

More specifically - 
I think the logic to replace a “Bundle-valued NoConn”, such as m.p in that example, would all live in `replace_noconn` here:

https://github.com/dan-fritchman/Hdl21/blob/09668d735cf692f6deb3de2d1ecd8324ec45db30/hdl21/elab/elaborators/portrefs.py#L260

That would require checking the `bundle_ports` as well as just `ports` of the target module `mod` here:

```python
        # Get the target Module's port-object corresponding to `portref`
        mod = portref.inst._resolved
        port = mod.ports.get(portref.portname, None) ## <- searching `mod.ports` here; also needs to search `bundle_ports`
```

From there, it looks like the `copy_port` function knows how to copy Bundle-valued ports. (And if it doesn’t, it’ll need to!)

https://github.com/dan-fritchman/Hdl21/blob/09668d735cf692f6deb3de2d1ecd8324ec45db30/hdl21/elab/elaborators/portrefs.py#L243

ThomasPluck, 2023-01-04:
Yes, but `bundle_ports` aren't automatically populated the same way that `ports` are for modules.

dan-fritchman, 2023-01-04:
> Yes, but bundle_ports aren't automatically populated the same way that ports are for modules.

Sure they are!  
Maybe this is a bit of confusion about what makes a Bundle port-visible.  
E.g.: 

```python
import hdl21 as h

@h.module
class M2a:
    p = h.Diff()

@h.module
class M2b:
    p = h.Diff(port=True)

print(M2a.bundle_ports)
print(M2b.bundle_ports)
```

Yields: 

```
{}
{'p': BundleInstance(name=p of=Bundle(name=Diff))}
```

Having port-visibility is an attribute of a `BundleInstance`. Particularly a boolean-valued one, named `port`.  
This mirrors the difference between the `Port` and `Signal` "types" - which is, there really is only one type. All of `Port`, `Input`, `Output`, etc are convenience functions which return `Signal`s, with an appropriate `SignalVisibility` attribute. 

Related: https://github.com/dan-fritchman/Hdl21/issues/39

ThomasPluck, 2023-01-06:
Should `bundle_ports` presentation be default (ie. no args required) - given that this is the desired cue that leads to a simple `replace_noconn` solution?

dan-fritchman, 2023-01-07:
OK I think we've just been misunderstanding each other a bit here. 

[6da279e](https://github.com/dan-fritchman/Hdl21/commit/6da279eae510fd598565eb7d956a1f5061683629) has the changes to enable `NoConn`s to connect to `Bundle`-valued ports. The "fix" is really all this line: 

https://github.com/dan-fritchman/Hdl21/blob/6da279eae510fd598565eb7d956a1f5061683629/hdl21/elab/elaborators/portrefs.py#L265

The rest is testing and the like. If you've got other test cases - let's roll them in!  
That change is layered atop some other things I'd done of late on #35, but I think should work stand-alone. 

--- 

Said change is really only one part of the issue scope here.  
Reaching back to the original bullet list: 

* Should they be allowed to connect non-unity-width ports?
  * Done. This... already appeared to work actually? This change tests it and succeeds. 
* Should they be allowed to connect to Bundle valued ports?
  * Done as of this change
* Should they be allowed in Concat?
  * Intent is "yes", implementation pending 
* Should they be allowed in AnonymousBundle?
  * Intent is "yes", implementation pending 
* Should they be Slice-able?
  * Intent is "no", and that intent is being met

So of those, there are two "pending": 

* `NoConn` in `Concat`
* `NoConn` in `AnonymousBundle`

We can make those separate issues, or continue keeping tabs on them here. I'd probably recommend the latter.