dan-fritchman/Hdl21 #44: Subtle Change in Netlisting

dan-fritchman, 2022-10-14:
We have a unit test `test_spice_netlister`, which checks text netlist output to look like this: 

```
        .SUBCKT DUT 
        + a_4 a_3 a_2 a_1 a_0 b_4 b_3 b_2 b_1 b_0 
        + * No parameters

        rres 
        + a_0 b_0 
        + 10000.0 
        + * No parameters

        ccap 
        + a_1 b_1 
        + 1e-11 
        + * No parameters

        lind 
        + a_2 b_2 
        + 1e-08 
        + * No parameters

        .ENDS
```

It changed to this: 

```
.SUBCKT DUT 
+ a_4 a_3 a_2 a_1 a_0 b_4 b_3 b_2 b_1 b_0 
* No parameters

rres 
+ a_0 b_0 
+ 10000.0 
* No parameters

ccap 
+ a_1 b_1 
+ 1e-11 
* No parameters

lind 
+ a_2 b_2 
+ 1e-08 
* No parameters

.ENDS
```

That change may be hard to spot - it's to these lines: 

```
        + * No parameters
```

Particularly, the `+` continuation got dropped.  
Now there happens to be nothing after them on the instance declaration, and they only write comments. So the "semantic netlist" remains the same. But I can't really see why this is happening.

aviralpandey, 2022-10-15:
This was an edit made by me in a branch that I didn't think was merged yet. 

The reason for this edit is that some simulators (ahem NGSPICE) don't like a line continuation that does not actually contain anything. Functionally, I don't think any of the other simulators care about this (atleast SPECTRE does not), which is why the change was in the upstream generic netlister, not the NGSPICE specific netlister.

dan-fritchman, 2022-10-15:
Ah yes - I forgot I had your feature branch of Vlsir!

dan-fritchman, 2022-10-19:
I suppose the other question here is _should_ HDL21 have any specific netlist-strings to test against. 
Maybe we should check against the VLSIR schema instead, and just see that it *can* netlist.  
Given a fixed version of VLSIR, not "whatever topic branch shows up in the environment", the specific netlist tests should continue to work. Question is whether they add value.

dan-fritchman, 2022-10-20:
@aviralpandey said he agreed with that last comment

dan-fritchman, 2022-11-29:
We didn't totally rip the netlisting out (comparing against VLSIR protobuf literals is even more verbose). But scaled back its strictness quite a bit. 

Maybe that change should go further, but for now, it works.

closed 2022-11-29 by dan-fritchman