dan-fritchman/Hdl21 #49: Bring `Poetry` Back!

dan-fritchman, 2022-10-17:
@vighneshiyer pointed out (some time ago now) that `Poetry` now supports the feature we really wanted it to, when we ditched it: editable dev-mode installs. 

I think it's worth bringing back, when we have the chance.

dan-fritchman, 2022-12-15:
I can also say firsthand, having used this on some other projects now, that Poetry does all the stuff we want for this project.  
Should be a relatively quick replacement of all the (4-5) `setup.py` files.

dan-fritchman, 2023-01-05:
This should probably more accurately be titled:  
Bring `pyproject.toml` back!

Since nowadays one can run `pip install -e` on projects with a TOML config (and no setup.py), without needing Poetry.

uduse, 2023-01-05:
I used poetry before and the editable dev install feature was the thing I wanted the most too :)))

dan-fritchman, 2023-01-06:
One complication using the `pip`/ `pyproject.toml` combo: it doesn't appear to support the `dev-dependencies` feature.  
Tried this elsewhere to no luck, and found many issues around the internet to this effect:  
https://github.com/python-poetry/poetry/issues/3348

In `setup.py` the development-dependencies are an "extra feature", which just happens to be named `dev`. Tried that too, to no avail.

dan-fritchman, 2023-06-19:
So two new notes here: 

* We at Hdl21 HQ are now playing with another project which relies on old-enough `pip` that at least *some* of our target `pyproject.toml` features won't work. And I think TOML just won't work period.  

* #149 started with a comment that "the setup.py method is now deprecated".  

It's not.  
The setuptools docs (linked in #149) say: 

> While it is perfectly fine that users write setup.py files to configure a package build (e.g. to specify binary extensions or customize commands), on recent versions of setuptools, running python setup.py directly as a script is considered deprecated. This also means that users should avoid running commands directly via python setup.py <command>.

So the deprecated thing is *running* `python setup.py <command>`.  
Which we don't do; in all cases invocation of that `setup` script (or at least advice on how to do so) comes though `pip`.

PR #219 (2024-05-24): Specify package Metadata using a pyproject.toml file
Closes #218 and closes #49 

## Updates
- I updated all the packages to specify their metadata in a [pyproject.toml](https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#writing-your-pyproject-toml) which replaced all setup.py files.

- Updated the package build backend to [flit ](https://flit.pypa.io/en/stable/)instead of using [build](https://build.pypa.io/en/stable/) as the package build back-end.

- Updated the sitepdks package name from "sitepdks" to "hdl21-sitepdks"

- The Pytest configuration was merged into the [pypr [... cut]
closed 2024-05-24 by dan-fritchman