dan-fritchman/Hdl21 #58: Improve this generator error message

dan-fritchman, 2022-11-06:
Report that this error: 

```
RuntimeError: Invalid generator call signature for create_module: [<Parameter "amp_specs">, <Parameter "gen_params">, <Parameter "amplifier_op">, <Parameter "tail_op">]
```

Was not sufficiently clear.  
And I agree. It should at least say what the *right* call-signature would be. E.g. 

```
RuntimeError: Invalid generator call signature for generator function `create_module`. 
  Generators must be of the form `def GenFunc(params: Params) -> Module`, where `Params` is a `paramclass`
  `create_module` has the arguments: [<Parameter "amp_specs">, <Parameter "gen_params">, <Parameter "amplifier_op">, <Parameter "tail_op">]
```

dan-fritchman, 2022-11-06:
@cem-yalcin how's that version look?

cem-yalcin, 2022-11-06:
Looks great!

dan-fritchman, 2022-11-29:
In #67

closed 2022-11-29 by dan-fritchman