dan-fritchman/Hdl21 #65: Allowing spice primtives to be defined directly from Hdl21

aviralpandey, 2022-11-15:
So far we support external modules with parameters, those parameters just have to be key-value only. 

I propose we allow external modules to optionally specify their parameters text as they need to be netlisted, and their spice prefix. This will essentially turn all of our primitives into external modules, which *could* be simpler than our current approach of writing down how to netlist every spice primitive we could ever run into.

So the new external module would look like:
```
message ExternalModule {
  // Qualified External Module Name
  vlsir.utils.QualifiedName name = 1;
  // Description
  string desc = 2;
  // Port Definitions
  // Ordered as they will be in order-sensitive formats, such as typical netlist formats. 
  repeated Port ports = 3;
  // Signal Definitions, limited to those used by external-facing ports.
  repeated Signal signals = 4;
  // Params
  repeated vlsir.utils.Param parameters = 5;
  // Spice types
  optional string spicetype = 6;
  // Spice Parameter netlist
  optional string paramnetlist = 7;
}
```

The netlister would then behave the same way it has, but instead of trying to netlist the parameters by mapping them itself, it would simply write the passed paramnetlist.

dan-fritchman, 2022-11-22:
This was a central topic in a recent meeting.  
Despite not thinking so at the time - I now think we probably want to have the "full-blown" escape hatch: a `Literal` type in both `hdl21.Module` and `vlsir.circuit.Module`.

dan-fritchman, 2023-05-23:
`vlsir.ExternalModule` and `hdl21.ExternalModule` both now include a `SpiceType` enum-valued field which should support this. 

And if not, re-open away.

closed 2023-05-23 by dan-fritchman