dan-fritchman/Hdl21 #73: Reduce confusion between `h.Module` and `@h.module`

aviralpandey, 2022-12-01:
I spent about an hour debugging this error that's hard to see with your eyes. 

We do generators like the following: 
```
def my_generator(p: MyParamType) -> h.Module:
  @h.module
  class MyModule:
    pass
  return MyModule
```

If you're fingers aren't great at typing like me, you'll sometimes type the following: 
```
def my_generator(p: MyParamType) -> h.Module:
  @h.Module
  class MyModule:
    pass
  return MyModule
```

This breaks the generator, and unfortunately the error is incredibly hard to decipher. 
We could try to make the error cleaner, but I want to offer an alternative. Let's rename the decorator to `@defmodule` instead. This way, the two are differentiated better. 

We can offer this alternate definition of the decorator initially and eventually phase out the old one.

dan-fritchman, 2022-12-01:
This is also #6. 
And I "fixed" - or at least added some code to handle - it in In https://github.com/dan-fritchman/Hdl21/commit/cbe66d8b991aacd219291c76eb376b258c3f8345 & https://github.com/dan-fritchman/Hdl21/issues/53.

The intent is that you get this error here: 
https://github.com/dan-fritchman/Hdl21/blob/945aa6eaf18696e29f79a262e53fe226735b7dca/hdl21/module.py#L63

```python
f"Module name {name} is a class. Did you mean to use the `module` decorator?"
```

What's the error you're getting?  
And would that seem to be clearer?

aviralpandey, 2022-12-01:
The error that I get is the following: 
```
Traceback (most recent call last):
  File "design_level_shifter.py", line 13, in <module>
    from circuits import *
  File "/tools/C/pandey/aura/modeling/circuits.py", line 121, in <module>
    class Inv:
TypeError: __init__() takes 1 positional argument but 2 were given
```

I did not see #6, but it seems that the error is still present. 

Taking a look at `hdl21/hdl21/module.py`, I don't think this catches even the basic use case.

```
 def __init__(self, *, name: Optional[str] = None):
        if name is not None and not isinstance(name, str):
            # Something wrong with this `name`.
            # Most common case: confusion with the `module` decorator.
            if isclass(name):
                msg = f"Module name {name} is a class. Did you mean to use the `module` decorator?"
            else:
                msg = f"Invalid Module name {name}, must be a string"
            raise TypeError(msg)
        self.name = name
```
Since  the `__init__` has the `*` limiting to only take one positional argument. We should probably remove that `*`? 

I get that we could fix the error handling, but is there a good reason to keep the decorator and class names the same with just a capitalization difference?

dan-fritchman, 2022-12-01:
FWIW there is a unit test for this - 
https://github.com/dan-fritchman/Hdl21/blob/945aa6eaf18696e29f79a262e53fe226735b7dca/hdl21/tests/test_hdl21.py#L707

... but it doesn't look like it's good enough.  
The `_init_ takes 1 positional arg` gobbledegook is a `TypeError` too. 

And you're right, it's the "no positional arguments allowed" `*` that screws it up. The decorator-syntax calls by position, and you get the gibberish error you reported. 

On the naming - I'm quite strongly of the opinion that those are the best two names available. That opinion can change, but is not presently terribly close to doing so. I'd much rather clear up the error message. 

Anyway here's an update that (I think) makes this clear: 
https://github.com/dan-fritchman/Hdl21/commit/0387364de335d0c5501d0a782aca3d3d86325a9e.  
Currently head of the `dev` branch, if that helps.

aviralpandey, 2022-12-01:
Seems to give the reasonable response: 

```
Generator(name=inv) raised an exception: 
Module name <class 'circuits.inv.<locals>.Inv'> is a class. Did you mean to use the `module` decorator?
```

dan-fritchman, 2022-12-01:
Beautiful! 

So we have it here: this should get merged with PR #74.