dan-fritchman/Hdl21 #81: Allow zero-arg `h.generator`

dan-fritchman, 2022-12-20:
Currently to make an argument-less `Generator`, you need to actually declare a parameter and give it a type. Often `h.HasNoParams` is used for the latter, like so: 

```python 
@h.generator
def PmosIdac(_: h.HasNoParams) -> h.Module:
  # ...
```

The issue: we should be able to just write 

```python 
@h.generator
def PmosIdac() -> h.Module: # <= here
  # ...
```

We do enough inspection of the `Generator` call-signature already, and lots of tuning when we invoke them during elaboration. This should be a pretty straightforward case to add.

dan-fritchman, 2022-12-20:
You might ask -  
but **why** make a zero-argument generator?  
Isn't that just a `Module`? 

Kinda yeah.  
I nonetheless end up doing it all the time, for a few reasons: 

* Over the course of development, you may go back-n-forth as to whether a given circuit has parameters or not. And the syntax for instantiating a `Generator` vs a `Module` differs. The former is generally "two calls", one for parameters and one for connections. The other is a single call with connections. Since adding the inline-`paramclass` generation, calling  zero-parameter generators such as `PmosIdac` above works without providing any arguments, e.g. `PmosIdac()(conns)`. This remains true if the generator has parameters, but you want their default values. 
  * So, in short, it can be easier to go back and forth between "some parameters" and none. 
* Perhaps less obviously - `Module`s "run" at import-time, and `Generator`s run at elaboration time. 
  * It's also not immediately obvious which is *better* - until building a sufficient-size project, and trying to import some small fraction of it. Module execution should never be _too_ bad, but does get noticeable. 
  * Related: many of the features tied up in #70, primarily for `Bundle`-valued ports, currently work only as `Generator`s. This is generally because generators have a bunch of caching-stuff built in, which allows them to avoid errantly trying to re-double-expand those Bundles.

ThomasPluck, 2023-01-17:
Another related piece of functionality that could be useful is arbitrary argument generators. That is, a way for generators to take ordinary arguments and have these become a "virtual" `paramclass` instance at Elaboration time that isn't immediately obvious to the user, ie.

```python
@h.generator
def myFirstGenerator(params : MyParams) -> h.Module:
    # ...
```

Could be written as:

```python
@h.generator
def myFirstGenerator(arg1 : int, arg2 : str, arg3 : Prefixed) -> h.Module:
    # ...
```

Maybe, this could be delivered alongside zero-arg generators?