dan-fritchman/Hdl21 #87: Connection to instances not accessible from the instance itself

uduse, 2023-01-03:
```python
@h.module
class Source:
    vin, vout = h.Ports(2)

@h.module
class Target:
    vin, vout = h.Ports(2)
    
@h.module
class Cat:
    vin, vout = h.Ports(2)
    
    source = Source(vin=vin)
    target = Target(vin=source.vout, vout=vout)
```

`print(Cat.instances['source'].conns)`

This gives
```
{'vin': Signal(name='vin', width=1, vis=<Visibility.PORT: 1>, direction=<PortDir.NONE: 3>, desc=None)}
```

Because the other connection is stored in `Cat.instances['target'].conns`, which prints
```
{'vin': PortRef(inst=Instance(name=source of=Module(name=Source)), portname='vout'),
 'vout': Signal(name='vout', width=1, vis=<Visibility.PORT: 1>, direction=<PortDir.NONE: 3>, desc=None)}
```

@dan-fritchman mentioned this connection is resolved in the elaboration process, but I am not sure how that works. Some elaboration on the elaboration process would be much appreciated.

I am currently solving this problem by always explicitly defining the connections as signals. For example, 

```python
@h.module
class Buffer:
    vdd, vss, vin, vout = h.Ports(4)
    n0 = h.Signal()
    inv_1 = Inverter(vdd=vdd, vss=vss, vin=vin, vout=n0) 
    inv_2 = Inverter(vdd=vdd, vss=vss, vin=n0, vout=vout)
```

uduse, 2023-01-03:
Related to #86 since I had problem walking in the module and gather all connections correctly.

dan-fritchman, 2023-01-03:
By adding a call to `h.elaborate` on `Cat`:

```python
import hdl21 as h 

@h.module
class Source:
    vin, vout = h.Ports(2)

@h.module
class Target:
    vin, vout = h.Ports(2)
    
@h.module
class Cat:
    vin, vout = h.Ports(2)
    
    source = Source(vin=vin)
    target = Target(vin=source.vout, vout=vout)

h.elaborate(Cat) # <- the one difference right here 
print(Cat.instances['source'].conns)
```

You get this: 

```
{
    'vin': Signal(name='vin', width=1, vis=<Visibility.PORT: 1>, direction=<PortDir.NONE: 3>, desc=None), 
    'vout': Signal(name='source_vout', width=1, vis=<Visibility.INTERNAL: 0>, direction=<PortDir.NONE: 3>, desc=None)
}
```

Which I think is what you want(?). 

Adding the discrete `Signal`s would work too, as long as you're diligent to *always* add them. But running through elaboration will always resolve all those port-references (along with lots of other details), or it'll fail if it detects problems. 

---

If the term "elaboration" is not familiar, it's a common one for dedicated HDLs (e.g. Verilog, VHDL). Think of it as "in-memory design expansion". Or think of it as the Hdl21 "compiler". Many, many things in Hdl21 as lazily evaluated at elaboration-time. These lazy-evals include: 

* Expanding Bundles into constituent Signals 
* Checking whether or not the width of connections match those of connected ports
* Turning "implicit Signals", generally from port-references e.g. the `PortRef` in your example, into concrete Signals like `source_vout`
* Probably most prominently: running all the Generator functions! 

Most things you'll ultimately want to _do_ with `hdl21.Module`s tend to invoke elaboration inline, including: 

* Converting to VLSIR protobufs
* Exporting to netlists
* Running spice-class simulations 

(Really all of those cascade through one another)

If you want to do some in-memory analysis as you noted - such as the "compute all the MOS areas" use-case mentioned elsewhere - you'll have to explicitly call `h.elaborate` yourself.

dan-fritchman, 2023-01-03:
BTW, one more shorthand:  
All of the `Module`-attributes (instances, signals, etc.) are dot-accessible by name.  
So stuff like: 

```
print(Cat.source is Cat.instances['source'])
```

Prints `True`.

uduse, 2023-01-04:
@dan-fritchman Thanks for the guidance. This is super helpful. 🙇‍♂️

closed 2023-01-04 by uduse