dan-fritchman/Hdl21 #91: Primitives can't be instantiated with `name` arg.

uduse, 2023-01-16:
```python
h.Nmos(name='nmos')
```
gives
```python
RuntimeError: Invalid call to `Primitive(name='Mos', desc='Mos Transistor', port_list=[Signal(name='d', width=1, vis=<Visibility.PORT: 1>, direction=<PortDir.NONE: 3>, desc='Drain'), Signal(name='g', width=1, vis=<Visibility.PORT: 1>, direction=<PortDir.NONE: 3>, desc='Gate'), Signal(name='s', width=1, vis=<Visibility.PORT: 1>, direction=<PortDir.NONE: 3>, desc='Source'), Signal(name='b', width=1, vis=<Visibility.PORT: 1>, direction=<PortDir.NONE: 3>, desc='Bulk')], paramtype=<class 'pydantic.dataclasses._Pydantic_MosParams_187650411960496'>, primtype=<PrimitiveType.PHYSICAL: 'PHYSICAL'>)` with arguments `{'name': 'nmos'}`: 

_Pydantic_MosParams_187650411960496.__init__() got an unexpected keyword argument 'name'
```

Is this a bug? or a feature that prevents users to set names explicitly?

dan-fritchman, 2023-01-16:
Not sure what gives the impression that it should work!  
The same thing fails for a `Module` and for a `Generator`: 

```python
import hdl21 as h

@h.module
class M:
    ...
M(name="whatever")  # fail

@h.generator
def G(_: h.HasNoParams) -> h.Module:
    return h.Module()
g = G()(name="whatever")  # also fails

nmos = h.Nmos(name="whatever")  # also fails
```

The semantics of all three are: 
* (a) Calls create `hdl21.Instance`s
* (b) Calls also *connect* into those instances

And `name` is not a connection. The error for the `Module` case better states this: 

```python
  File "/Users/dan/dev/Vlsir/Hdl21/hdl21/instance.py", line 112, in connect
    raise TypeError(f"{self} attempting to connect non-connectable {conn}")
TypeError: Instance(name=None of=Module(name=M)) attempting to connect non-connectable whatever
```

If you do want to create an instance of `h.Nmos`, there are two principal ways to do so - 

(a) "The usual way":

```python
@h.module
class M:
    n1 = h.Nmos()() # Module-magic sets `name` to `n1`
```

Here the second call creates the `Instance`, and the `Module` setattr-magic sets its name to `n1`. 

(b) Create the `Instance` explicitly. That passage above is equivalent to: 

```python
@h.module
class M:
    n1 = h.Instance(of=h.Nmos()) 
```

or: 

```python
@h.module
class M:
    ...
m.add(h.Instance(of=h.Nmos(), name='n1')) 
```

uduse, 2023-01-23:
I got the impression that this could work because I thought names could be set similar to the port connections:

```python
import hdl21 as h

vss = h.Port()
nmos = h.Nmos()(s=vss)  # this works
nmos.s = vss  # this also works
nmos.name = 'nmos_name'  # this actually assigns the name correctly
```

dan-fritchman, 2023-01-25:
OK great I see. On those questions: 

> Is there a way to create an instance with a specific name without nesting it inside of a module? 

Yep. In fact it's included in one of the examples above! 

```python
m.add(h.Instance(of=h.Nmos(), name='n1'))  ## <= here

# more verbosely: 
i = h.Instance(of=h.Nmos(), name='n1')  ## <= `i` is an `Instance`, with a name
m.add(i)
```

Create the `Instance` explicitly, and it'll take `name` as a constructor keyword argument.  
Perhaps notably `Module.add` also accepts a `name` function argument, and applies it to "unnamed" module-attributes.

> or an instance should always be inside of a module and never be used alone?

I mean their *utility* is awfully limited outside of a parent `Module`. Curious what you would imagine doing with such a thing. Making it a generator parameter maybe, to later be inserted in a Module?  
Nonetheless they can, and in all cases at least briefly do, exist without parent Modules, such as `i` in the excerpt above.

dan-fritchman, 2023-01-25:
The "renaming" case actually highlights a potential problem, and motivated filing #94.  
As it says, for now once you've added things to a parent `Module`, "don't do that".

uduse, 2023-01-25:
> Yep. In fact it's included in one of the examples above!
Sorry, stupid me completely missed that 😢

> Curious what you would imagine doing with such a thing.
Nothing specific, was just trying to do some day-to-day Python manipulation of `Mos` objects on the instance level. e.g., my recent PR #86 , but ended up working around it.

> "don't do that"
Yes, sir 🫡

closed 2023-01-25 by uduse