dan-fritchman/Hdl21 #97: `h.Module` can be serialized but not deserialized

uduse, 2023-02-01:
```python
import pickle
import hdl21 as h

m = h.Module(name="Rick")
s = pickle.dumps(m)
pickle.loads(s)
```

This script gives:
```
---------------------------------------------------------------------------
UnpicklingError                           Traceback (most recent call last)
.../sandbox.py in line 7
      184 m = h.Module(name="Rick")
      185 s = pickle.dumps(m)
----> 186 pickle.loads(s)

UnpicklingError: state is not a dictionary
```

My guess is that we need to inject more class magic to support pickling for `h.Module`?

Related: #93

uduse, 2023-02-01:
It seems the problem is that this method overwrites `__getstate__` but doesn't provide a correspoding `__setstate__` https://github.com/dan-fritchman/Hdl21/blob/49c07a181c203d64e4739abeca2be59adbd76a2c/hdl21/qualname.py#L39-L63

dan-fritchman, 2023-02-01:
...Yeah it's not a `__getstate__` that's actually gonna *work*, in the sense of recovering the module content. It was at some point used for generated-module naming and/or parameter-izing. It probably should be removed.  

There's a bit on the history of trying to pickle these things in here: 
https://github.com/dan-fritchman/Hdl21/issues/93

Short answer is: there's never really been an intent to, or use case for, pickling these things?  
As you said, VLSIR is the primary intended serialization mechanism. And `hdl21.Module` does include a handful of things, especially in the background, which are not pickle-able (e.g. its source Python module). So it would require some custom pickle-ization. 

If that's a worthwhile feature, have at it! But I wouldn't anticipate doing so otherwise.

uduse, 2023-02-01:
> Short answer is: there's never really been an intent to, or use case for, pickling these things?

I am using Ray to parallelize things and `h.Module` is something I need to passaround. I thought Python native serialization (pickle) is supported out-of-the-box through VLSIR so I was surprised it didn't work. 

I am planning to just do VLSIR manually for now, until it gets painful enough then I might do a PR 🤣