dan-fritchman/Hdl21 #107: SKY130 PDK Model Loading Techniques

ThomasPluck, 2023-04-01:
From the discussion had in #103 and [externally with ngspice people](https://sourceforge.net/p/ngspice/discussion/120973/thread/bae9b098ef/), it seems as if ngspice loading of SKY130 can't be improved without serious changes to either ngspice or the SKY130 PDK itself.

So without a simple setting, the next best thing would be to improve how HDL21 Sims construct their SPICE files for external PDKs, how this could work:

1. Add a new "automated dependencies" function to the SKY130-HDL21 pdk-module.
2. The arguments of this automated dependency function would be the `Sim` itself and return the `Sim` with added `Lib` attributes
3. The class will then use the testbench to determine what modules are required and find the appropriate .model in `libs.ref`.
4. Then the directory + appropriate models (because that is a thing in SKY130) tacked onto the `Sim` using procedural syntax.

This is unique to the PDK, so it will be written within the pdk-module and shouldn't touch HDL21 proper or VlsirTools. Probably the largest complication here, is that it'll be necessary to build out a JSON that will be look-up table for which model, in which corner, for which model, etc.

dan-fritchman, 2023-04-03:
That is possible, although - 

* (a) Check out the feedback here: https://github.com/dan-fritchman/Hdl21/issues/104#issuecomment-1494984447 
  * While it's not exactly the same mechanism I remembered, that definitely does (or at least has) had the desired effect of speeding up ngspice start-up by like tens of seconds. Maybe fast enough without adding all these mechanics. 
* (b) If doing this "on-demand loading", would it be worth proposing as a PR to ngspice instead?

ThomasPluck, 2023-04-04:
Thanks for clarifying! I have a commit lined up that has this feature alongside setting "nomodcheck" to be true.

I agree, it's a monkey patch. NgSpice is really where this needs to be fixed for everyone. I'll open up an issue and see where it goes.

ThomasPluck, 2023-04-16:
Disbanded in handling #111 not a feature that will be supported for an evolving PDK and user-base.

closed 2023-04-16 by ThomasPluck