dan-fritchman/Hdl21 #112: Jupyter Notebook Compatability Issues

ThomasPluck, 2023-04-18:
I tried simulating a SKY130 circuit in Jupyter notebook and I get the following traceback:

```python
Output exceeds the size limit. Open the full output data in a text editor
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Cell In[3], line 34
     25     inc = sky130.install.include(h.pdk.Corner.TYP)
     28 opts = vsp.SimOptions(
     29     simulator=vsp.SupportedSimulators.NGSPICE,
     30     fmt=vsp.ResultFormat.SIM_DATA,
     31     rundir="./scratch"
     32 )
---> 34 results = Sim.run(opts)

File ~/Projects/HW-Design/Hdl21/hdl21/sim/data.py:387, in Sim.run(self, opts)
    385 def run(self, opts: Optional[vsp.SimOptions] = None) -> vsp.SimResultUnion:
    386     """Invoke simulation via `vlsirtools.spice`."""
--> 387     return run(self, opts=opts)

File ~/Projects/HW-Design/Hdl21/hdl21/sim/data.py:411, in run(inp, opts)
    407 from .to_proto import to_proto
    409 inp.Tb.props.set("simulator", opts.simulator.value)
--> 411 return vsp.sim(inp=to_proto(inp), opts=opts)

File ~/Projects/HW-Design/vlsir/VlsirTools/vlsirtools/spice/spice.py:85, in sim(inp, opts)
     77 def sim(
     78     inp: OneOrMore[vsp.SimInput], opts: Optional[SimOptions] = None
     79 ) -> OneOrMore[SimResultUnion]:
...
     34         "asyncio.run() cannot be called from a running event loop")
     36 if not coroutines.iscoroutine(main):
     37     raise ValueError("a coroutine was expected, got {!r}".format(main))
```

Fixing this requires Jupyter notebooks knowledge than I currently have.

uduse, 2023-04-19:
#78 should have fixed this, use `run_async` instead.

ThomasPluck, 2023-04-19:
That works great, thanks!

closed 2023-04-19 by ThomasPluck

dan-fritchman, 2023-04-19:
Thanks Zeyi!  
Reminds me I had been meaning to file https://github.com/dan-fritchman/Hdl21/issues/113