dan-fritchman/Hdl21 #113: Double-Awaiting Sim Results

dan-fritchman, 2023-04-19:
We got something wrong piping `await`s around the asynchronous sim launchers.  
Currently if called in an async context like below, you need to `await` them twice: 

```python
async def some_async_caller_function():
    """ An example of calling `run_async` from another async function... and it not going as expected """

    @h.sim.sim
    class MySim:
        ... 

    # Run the sim
    somehow_this_is_a_coroutine = await MySim.run_async()

    ## FIXME: why on earth do we have to double-await this thing?!?
    ## Get some better tests in async-land @dan_fritchman
    sim_result = await somehow_this_is_a_coroutine

    # Now THAT is a `SimResult`
    do_stuff_with_results(sim_result)
```

uduse, 2023-04-19:
https://github.com/dan-fritchman/Hdl21/blob/3bb45f6e5c30cfc8ebf3ae57e2185f44d7c06eb5/hdl21/sim/data.py#L393

-> ` return await run_async(self, opts=opts)` I guess?

dan-fritchman, 2023-04-28:
Yep I believe that'd work.  
Also believe dropping the `async` modifier from any functions that don't actively `await` anything - such as the `run_async` method highlighted above - would work too.  
As in this simple example: 

```python
import asyncio
from typing import Awaitable

async def inner() -> Awaitable[int]:
    await asyncio.sleep(0.001)
    return 11

def middle() -> Awaitable[int]:
    return inner()

def outer() -> Awaitable[int]:
    return middle()

x = asyncio.run(outer())
print(x)
```

I (as evidenced here) don't use the async-ecosystem much. Do you think it's preferable to propagate the `async def`s and `return await`s through any callers? I suppose it makes its a bit more clear that's what's happening under the hood (although hopefully the `_async` function-name does too).

uduse, 2023-04-29:
I think using `async def` is a better practice. The `def` version can't really be used as a vanilla function and its return value always has to be awaited to be useful anyways.

e.g.,
```python
import asyncio
from typing import Awaitable

async def inner() -> Awaitable[int]:
    await asyncio.sleep(0.1)
    return 1

def middle() -> Awaitable[int]:
    return inner()

async def middle_async():
    return await inner()
    
async def outer():
    x = middle()
    print(x)  # feels like printing a real value, but it's not
    
    y = await middle()
    print(y)  # the real way to actually print the value
    
    z = await middle_async()
    print(z)  # the least surprising way to print the value

asyncio.run(outer())
```

Also the return value of `async def` functions should be type hinted without the `Await`, e.g.,
```python
async def show() -> int:
    return 5
```
Think the `async` in this case like a weirdly positioned decorator that modifies the entire `show()` function.

uduse, 2023-04-29:
One reason of us has to do weird async stuff here is because how problem in the Vlsir project.

Here:
https://github.com/Vlsir/Vlsir/blob/e7f5f91a0a4e14469cd33c6a5c3b9b1b321a6058/VlsirTools/vlsirtools/spice/spice.py#L77-L85

We should use `asyncio.get_running_loop()` to detect the running asyncio loop and only start a new one with `asyncio.run` if we don't have one. This way we can make sure `sim()` can always be called synchronously.

dan-fritchman, 2023-05-01:
OK so I take it - 

* (a) There are situations where one is running in an async context, but doesn't necessarily "control it", or really even necessarily know that's the case. Jupyter has come up as an example. 
* (b) The `asyncio` library only allows one event loop at a time(?). 
* (c) `asyncio.run` tries to create one, and fails if it can't, according to [this](https://docs.python.org/3/library/asyncio-runner.html#asyncio.run). 

Do I have (b) right?  
And if so, is there like a "use the existing event loop if there is one, or create one if there isn't" function in `asyncio`? Or does one need to check for themself? 

Also - looks like a high-value VLSIR PR!

uduse, 2023-05-01:
(a) Yes.
(b) You can do weird things with multiple threads each with a loop, but you shouldn't unless there's a strong justification. The better alternative is to do multi processes each with an event loop.
(c) I think so too.

> And if so, is there like a "use the existing event loop if there is one, or create one if there isn't" function in asyncio? Or does one need to check for themself?

We have `asyncio.get_event_loop()`, but based on the [docs here](https://docs.python.org/3.11/library/asyncio-eventloop.html#asyncio.get_event_loop) this API is discouraged for some reason and it seems the way to do it now is to try catch `asyncio.get_running_loop()` then `asyncio.new_event_loop()` if that didn't work.

dan-fritchman, 2023-06-05:
OK at least the smaller issue here of double-`await`ing, while we still have `async`, is done. 

#122 will be the real action.

closed 2023-06-05 by dan-fritchman