dan-fritchman/Hdl21 #135: Netlisting with `SpiceType`

dan-fritchman, 2023-06-07:
#134 adds this (failing) test: 

```python
def test_netlist_spicetypes():
    """# Test netlisting `ExternalModule`s with `SpiceType`s"""
    from hdl21.external_module import SpiceType
    from vlsirtools.netlist import NetlistFormat

    NmosModel = h.ExternalModule(
        name="NmosModel",
        port_list=deepcopy(h.Mos.port_list),
        paramtype=h.HasNoParams,
        spicetype=SpiceType.MOS, # <= **** THIS THE POINT RIGHT HERE ****** <=
    )

    @h.module
    class HasSpiceTypes:
        x = h.Signal()
        m = NmosModel()(d=x, g=x, s=x, b=x)

    h.netlist(HasSpiceTypes, sys.stdout, fmt=NetlistFormat.SPICE)
    h.netlist(HasSpiceTypes, sys.stdout, fmt=NetlistFormat.XYCE)
    h.netlist(HasSpiceTypes, sys.stdout, fmt=NetlistFormat.NGSPICE)
    h.netlist(HasSpiceTypes, sys.stdout, fmt=NetlistFormat.SPECTRE)
```

* Ngspice and Spectre formats throw exceptions
* Spice and Xyce formats run, but produce this output:

```
* Anonymous `circuit.Package`
* Generated by `vlsirtools.SpiceNetlister`
* 

.SUBCKT HasSpiceTypes
* No ports
* No parameters

mm
+ x x x x 
* No parameters

.ENDS

; Anonymous `circuit.Package`
; Generated by `vlsirtools.XyceNetlister`
; 

.SUBCKT HasSpiceTypes
; No ports
; No parameters

mm
+ x x x x 
+ ; No parameters

.ENDS
```

Both of which have a big problem - they don't say what kinda device to instantiate!  

The resolution here probably ultimately lies in `Vlsir`.  
But starting here, as the initial (and probably, most) test-cases are driven from Hdl21.

ThomasPluck, 2023-06-08:
I can't replicate the NgSpice error, but the netlists are still off. So, its really just a matter of non-subcircuit devices not having an extra line which states: I am device `x` - I guess you could just replicate how this is done for subcircuits in VlsirTools. I'll assign this one to myself and hopefully have a work around done by midnight in Dublin.

dan-fritchman, 2023-06-19:
Complete via #134

closed 2023-06-19 by dan-fritchman