dan-fritchman/Hdl21 #147: Refactor SKY130 Digital Cells to be in a Seperate Namespace

ThomasPluck, 2023-06-23:
Including the digital cells alongside the primitives has a few annoying side-effects:

- Long module loading time
- Confusion between primitives and cells
- Unnecessarily large/difficult to navigate PDK data files
- Very long and redundant cell names

It'd be better to have standard cell libraries in different namespaces for these reasons and make it something as slick as #114, where we can just use:

```python
from gf180 import digital7track as d
simple_and_gate = d.and2_1
```

Instead of:
```python
from sky130 import modules as s
# Now you wait for 5 whole seconds
simple_and_gate = s.sky130_fd_sc_hdll__and2_1
```

dan-fritchman, 2023-06-24:
Sounds great, totally agree.

ThomasPluck, 2023-06-25:
So getting into the details of this issue, it seems that with the `SimpleNamespace` approach:
```python
from sky130.digital import high_density as hd
```
Is actually logically equivalent to:
```python
import sky130.digital as _tmp
hd = _tmp.high_density
del _tmp
```
Which is... horrible. This stems from the fact that `SimpleNamespace` is a variable type rather than a submodule. To really get fine-grained control of imports and ensure that we're only importing the cells we're interested in, we'd have to split up the files into individual submodules which are populated without the use of `SimpleNamespace`.

I've tried this out and works pretty well on my local machine but the call changes to:
```python
import sky130.digital.high_density as hd
```
And teasing apart the primitives to mirror this change, would require effort that may not be worth it... hey, you might have 4s of unnecessary overhead when loading in your digital cells, but at least you don't have that overhead when loading in the base PDK module and the primitives (the core functionality of Hdl21).

So, going to leave this open for now, change the name to reflect the state of affairs, and mark it a "good first issue" for newcomers.

ThomasPluck, 2023-06-25:
Okay was just being lazy here this got resolved in latest push of #114

closed 2023-06-25 by ThomasPluck