dan-fritchman/Hdl21 #150: Release v4

dan-fritchman, 2023-06-24:
Plan: release version 4 this week or early next.  
Until then, planning to focus on: 

* (a) Any open bugs
* (b) Any "semi-bugs", i.e. common points of user confusion
* (c) Docs
* (d) Examples

To do: 

* Roll in pending PRs:
  * #114 / Issue #140 
  * #148
  * At least one to VLSIR, for the schema `literals`
* Documentation additions: 
  * #145 
  * #136 
  * #60 
  * Recent PDK additions
  * Lotta stuff about Bundles
* Also roll in: 
  * #139 
* Maybe roll in: 
  * #147 
  * #142
  * #120 
  * #30
* Examples:
  * #126 
  * #125 
* I think these are done(?): 
  * #68 
  * #39

curtisma, 2023-07-15:
@dan-fritchman Can you close this?  v4 was released.
Maybe start a v5 release issue to start.  Or make a v4.1 or v5 milestone?

dan-fritchman, 2023-07-18:
Didn't know we had so much housekeeping enthusiasm!  
A few reminders to future-selves as to what made it: 

* Partially in v4, remaining open to follow: 
  * #140 
  * #125 
* Didn't make it: 
  * #65 
  * #71 
  * #115 

Everything else in the list above is in v4.

closed 2023-07-18 by dan-fritchman