dan-fritchman/Hdl21 #153: PyPI Publish + Semver Actions

ThomasPluck, 2023-07-02:
A couple nice to have actions to keep the CI/CD buttery smooth:

- [ ] https://github.com/marketplace/actions/semver-release
- [ ] https://github.com/marketplace/actions/pypi-publish

dan-fritchman, 2024-05-24:
Having spent plenty of this morning publishing version 6, I can imagine plenty of improvements. 

Just which stuff do these actions do, though?  
My brief read of the docs is: 

- Publish to PyPi for semver-compatibily tagged (... somethings? PR names?)
- Don't modify any code, markup, etc that has those versions embedded

The reality of the headaches here are really that: 

- The "Hdl21 family" is kinda like an unofficial "workspace" including quite a few packages. Mostly python (Hdl21 itself, all the built-in PDKs, VlsirTools, etc), but also the VLSIR bindings to handfuls of other languages. 
- That "workspace" is not a formal/ technical thing in Python, or at least any Python tools I know. (Unlike, say, Rust, npm, etc.). And there's certainly no such good multi-language tool that I know, with commands e.g. "bump versions in all".
- So, we keep the same version-string in all of em. They all get published in lock step, whether or not they need to or have changes.
- There's just a lotta finding and replacing that version string? And sometimes missing some instances. 
- And then, in Python in particular, there's the painful culture of the runtime `__version__` variable, which is an extra pain to keep in sync with the install-time values in `setup.py` or `pyproject.toml`. (VLSIR's just failed to update between v2 and v5.)

dan-fritchman, 2025-02-08:
@hpretl - do you want to take a look at automating releases here? 

I'm all for a system like Thomas proposed, in which e.g. pushing a particular tag-format (usually `vX.Y.Z`) gets the CI/ actions server to auto-publish. But also don't have bandwidth to add such automation any time soon. Nor to quickly spin releases on demand.

hpretl, 2025-02-08:
@dan-fritchman I'll take a look whether we have the BW to add this.