dan-fritchman/Hdl21 #175: MOSFET Primitive `Enums` and `MosParams`

kcaisley, 2023-08-26:
I have a couple of questions/proposed changes regarding the `Enums` and `Params` for primitive transistors. Relevant snippet from `primitives.py`:

```python
class MosType(Enum):
    """# MOS Type (NMOS/ PMOS) Enumeration"""
    NMOS = "NMOS"
    PMOS = "PMOS"

class MosVth(Enum):
    """# MOS Threshold Enumeration"""
    STD = "STD"
    LOW = "LOW"
    HIGH = "HIGH"
    ULTRA_LOW = "ULTRA_LOW"
    ZERO = "ZERO"
    NATIVE = "NATIVE"

class MosFamily(Enum):
    """# MOS Family Enumeration"""
    NONE = "NONE"
    CORE = "CORE"
    IO = "IO"
    LP = "LP"
    HP = "HP"

@paramclass
class MosParams:
    """# MOS Transistor Parameters"""
    w = Param(dtype=Optional[Scalar], desc="Width in resolution units", default=None)
    l = Param(dtype=Optional[Scalar], desc="Length in resolution units", default=None)
    npar = Param(dtype=Scalar, desc="Number of parallel fingers", default=1)  # FIXME: rename
    mult = Param(dtype=Scalar, desc="Multiplier", default=1)
    tp = Param(dtype=MosType, desc="MosType (Nmos/ Pmos)", default=MosType.NMOS)
    vth = Param(dtype=MosVth, desc="Threshold voltage specifier", default=MosVth.STD)
    family = Param(dtype=MosFamily, desc="Device family", default=MosFamily.NONE)
    model = Param(dtype=Optional[str], desc="Model (Name)", default=None)
```

I understand the `MosParams.model` field is designed as an "escape hatch" to allow specifying less-common device flavor that don't warrant including in the `MosFamily`, `MosVth`, and `MosType` enums. (Leaned from Dicussion #103)

1. With that in mind, two of the PDKs I use have separate baseband and RF device device models, which share the same core BSIM device but include different degrees of LDE modeling. I am currently using the `model` escape hatch to toggle between the two when using `compile()` to convert primitives to `ExternalModule`s. But I'm curious if you feel this device flavor is common enough to warrant adding a `RF="RF"` member to the `MosFamily` enum?

2. Can I propose adding a `ULTRA_HIGH = "ULTRA_HIGH"` member to the `MosVth` enum?

3. Can I propose addressing the `# FIXME: rename` for `MosParams.npar` below, by renaming it to `MosParams.nf`?

Will open a pull request if these small changes are useful. (Would be sure to go through the examples, to update the references to `npar`.)

dan-fritchman, 2023-08-30:
Thanks for digging in here! 

1. Adding a `MosFamily.RF` variant makes sense to me
2. Is `ULTRA_HIGH` a real one you've encountered, or just, like, for symmetry with `ULTRA_LOW`? 
3. Go for it! There are `FIXME`s all over to move the `primitives.MosParams` field names to `nf` and `mult`. They just haven't happened since they require updating the downstream PDK packages, including the non-built-in ones. We can roll those in as they come up.

kcaisley, 2023-08-31:
Okay, great. I've made the changes for the enums, and renamed all instances of `npar` -> `nf` and all instances of `m`, `mf`, and `vm` -> `mult`. I've grouped them together with my outstanding PR #164 as all the changes are related to device `MosParams`.

Regarding `ULTRA_HIGH` Vth class devices: Yes, I know some PDKs that have them. One [publicly documented](https://europractice-ic.com/technologies/asics/tsmc/) is a TSMC 28nm PDK, which has ULVT, LVT, SVT, HVT, UHVT (and even an 'extremely high' EHVT variety, to boot). From the linked page:

[image]

Closing now (as this is implemented in #164 )

closed 2023-08-31 by kcaisley