dan-fritchman/Hdl21 #215: Python 3.12

dan-fritchman, 2024-03-26:
Not immediately working. E.g.: 

https://github.com/dan-fritchman/Hdl21/actions/runs/8443756151/job/23127988377

(Or at least our CI is not immediately working.)

dan-fritchman, 2024-08-15:
Noting https://github.com/Vlsir/Vlsir/pull/98 (basically the same topic for VLSIR) worked, without any real changes.  
Presumably just updates to our dependencies in the test environment.

joamatab, 2024-08-15:
Awesome!

could we make a new release?

dan-fritchman, 2024-08-15:
Should probably actually fix this one first.  
You wanna do a quick PR for it?

joamatab, 2024-08-30:
the tests worked for me

[image]

dan-fritchman, 2024-08-31:
So I don't see it passing anywhere on the CI runs?  
E.g. the most recent run:  
https://github.com/dan-fritchman/Hdl21/actions/runs/9786217659

We explicitly enable "continue on error" for py3.12 here:
https://github.com/dan-fritchman/Hdl21/blob/5633bc66b19dfecedbda23f69ec3e4818021afe2/.github/workflows/test.yaml#L43

Maybe just PR that out, and enough stuff "behind the scenes" will have changed to make it "just work".

dan-fritchman, 2024-09-06:
#232 "fixes" this.  
Which is to say, fixes our test setup.  
Particularly I split out the big matrix of Python and pydantic test-versions from the containerized simulation tests.

dan-fritchman, 2024-09-06:
Just in time for Python 3.13's release on October 1:  
https://peps.python.org/pep-0719/

dan-fritchman, 2024-09-06:
OK well spoke to soon, #232 doesn't quite work yet.  
Appears the combination of Python 3.12 & Pydantic 1.9 don't play nice. E.g.:
https://github.com/dan-fritchman/Hdl21/actions/runs/10733148162/job/29766001200

Noting for Python 3.7-3.8 it's "the other way around". Pydantic v1 works, v2 fails. 

I don't see what we're gonna be able to do about that, other than maybe writing a more elaborate dependency specification (if that's possible). Probably just like a note or table.